josefadamcik/SofleKeyboard #142: Sofle Choc ZMK Support

brianlow, 2022-04-01:
Goal: configure ZMK firmware to work with the Sofle Choc

Post your progress, findings and ideally a branch of your code even if not fully working

killua99, 2022-04-01:
Could we open a Threat on ZMK Discord and post here the relevant parts and have a more lose discussion on discord ?

brianlow, 2022-04-01:
Yeah for sure! I don't know ZMK and I'm not working on it so can't offer direct help. Hopefully there are others

infused-kim, 2022-04-01:
The existing zmk config works fine with the sofle choc. The only thing it had missing was the underglow. 

I created a PR for it here:
https://github.com/zmkfirmware/zmk/pull/1188

But zmk currently doesn’t fully support some of the more unique features of the sofle, such as displays and the split side encoder. 

I have created a fork that integrates fixes and some additional features here:
https://github.com/infused-kim/zmk/tree/sofle

And the easiest way to switch to that fork is to fork this zmk config:
https://github.com/infused-kim/zmk-config-sofle

And we briefly talked on Reddit about adding more info about wireless builds to the sofle build guide, which I started working on already. 

I am planning to also add info about zmk and the fork to it.

killua99, 2022-04-02:
One last comment, I'm on Sofle RGB V2.1 not the Choc perhaps that's why the RGB wont work.

infused-kim, 2022-04-02:
@killua99 according to the schematic the RGB sofle is using the same pins as the choc. So the zmk config should work without changes. 

Unless your vendor didn’t use the Gerbers published here. 

To debug this I would check:

* Is external power turned on? You can the external power toggle to your key map to test it. If you have displays and they work, then that’s also a sign that it’s on. 
* Also check if the backlight is actually on. Again, you  should add the toggle command to your key map. 
* Try if the backlight works while connected to USB. Maybe some power management is doing something. 

If it’s still not working, try flashing qmk to the primary side. 

Just make sure you don’t connect the TRRS cable while the nice!nanos are connected to the battery. 

If it doesn’t work with qmk either, then perhaps it’s a hardware issue.

brianlow, 2022-04-02:
@infused-kim thanks for posting all the info!

@killua99 the Sofle RGB uses the same pin but has a few differences
* adds underglow
* adds indicator led
* per-key LEDs are arranged in a different order 

So the Sofle Choc firmware should work but not all LEDs will light up (only first 58) and any color patterns won't look right.

killua99, 2022-04-03:
To debug this I would check:

> Is external power turned on? You can the external power toggle to your key map to test it. If you have displays and they work, then that’s also a sign that it’s on.

Yes, I'm compiling the firmware with your changes https://github.com/killua99/zmk-config-sofle and I can toggle the power with the adjust key combination. OLED stop working with I toggle off and working with on.

> Also check if the backlight is actually on. Again, you should add the toggle command to your key map.

Toggle command seems to be working, but no LED is one, nor indicator or underglow

> Try if the backlight works while connected to USB. Maybe some power management is doing something.

I did try that, I plugged USB and still the same no changes on my eyes.

Seems like I could have messed up the solder pins, and the vender said is the RGB V2.1 ([beekeeb](https://docs.beekeeb.com/build-guide/sofle-rgb-v2.1-soflekeyboard-build-log-guide-with-photos#rgb-leds-for-the-right-hand)) or the LEDs when I sold them I burn them directly. I've no more ideas.

infused-kim, 2022-04-03:
@killua99 the final thing I would try is flashing qmk. If it doesn’t work with it either, then you know it’s a hardware issue. 

I think the rgb has a few jumpers that need to be soldered. And if you have a multimeter you can check the led pins to see if they have power.

billyjacoby, 2022-09-02:
Hey, I'm trying to build this currently for adding RGB underglow to my lily58 and i'm seeing this error, any. thoughts?

`devicetree error: 'global-quick-tap' appears in /behaviors/hold_tap_tap_preferred in nice_nano_v2.dts.pre.tmp, but is not declared in 'properties:' in /__w/zmk-config-sofle-lily/zmk-config-sofle-lily/zmk/app/dts/bindings/behaviors/zmk,behavior-hold-tap.yaml
CMake Error at /__w/zmk-config-sofle-lily/zmk-config-sofle-lily/zmk-zephyr/cmake/dts.cmake:230 (message):
  gen_defines.py failed with return code: 1
Call Stack (most recent call first):
  /__w/zmk-config-sofle-lily/zmk-config-sofle-lily/zmk-zephyr/cmake/app/boilerplate.cmake:535 (include)
  /__w/zmk-config-sofle-lily/zmk-config-sofle-lily/zmk-zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:24 (include)
  /__w/zmk-config-sofle-lily/zmk-config-sofle-lily/zmk-zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:91 (include_boilerplate)
  CMakeLists.txt:17 (find_package)

-- Configuring incomplete, errors occurred!
FATAL ERROR: command exited with status 1: /usr/local/bin/cmake -DWEST_PYTHON=/usr/bin/python3 -B/__w/zmk-config-sofle-lily/zmk-config-sofle-lily/build -S/__w/zmk-config-sofle-lily/zmk-config-sofle-lily/zmk/app -GNinja -DBOARD=nice_nano_v2 -DZMK_CONFIG=/__w/zmk-config-sofle-lily/zmk-config-sofle-lily/config -DSHIELD=lily[58](https://github.com/billyjacoby/zmk-config-sofle-lily/runs/8163058298?check_suite_focus=true#step:9:59)_right`