josefadamcik/SofleKeyboard #145: Help: Lost on RGB matrix layout coordinates

ryuugaplays, 2022-05-02:
**Kind of issue?**
 Software / HELP!_

**Which Board?**
RGB 

**Which Board Revision?**
RGB =   2.1

**What steps have you taken**
* [ x ] I have searched the Issues page
* [ x ] I have read the documentation 
* [ x ] For LED issues I have checked the following things 
        - the pro micro is programmed (and currently telling the LEDs to do something)
        - the pro micro is powered (and showing LEDs)
        - the LEDs are powered (check 5v across pins 1 and 3 of J1)
        - are any of the LEDs showing 

**What is the issue?**
 Hello, after few weeks of assembly and troubleshooting I have finally put together my very own sofle!! 
currently trying to dig into the QMK code and edit stuff to my liking but have hit a wall trying to decipher the RGB matrix  coordinates. I have read through the docs and github but still feel lost. 
I have only the underglow and perkey lights. indicator lights is disabled.

So firstly I understand that {1, 6, hsv} means from LED 1, lightup 6 LEDs. However, according to [this](https://imgur.com/RtbLSeO) diagram, underglow for LED1 is actually 02
and then there is 35+1 what does that mean?

on QMK docs it says you can define the layout which is what is happening here on the right #define layout( 
can  i use this in the code so instead of {1, 6, hsv} I do {L01, 6, hsv} for example

I  guess the question here is, there are so many sets of numbers, What does the numbers in the code correspond to in terms of the physical location and how do I figure out which ones are the underglow and where does the perkey lights start and end?
 also btw is it normal that i have to flash both sides whenever I make changes to the RGBs? I tried flashing one side only but the other side remains unchanged

**What you expected:**
 LED1 is number 1

DaneEvans, 2022-05-02:
That depends.
What is your setup?

If you use the top indicator, then it is the first. If you bypass it, then
the first drop LED is the first.

The macros make it easier, but you can just count and do it manually

ryuugaplays, 2022-05-04:
So after some rigorous trial and error, flashing and reflashing, I had finally found the proper configuration for my LEDs. 
The issue was two folds: 

First, as mentioned above, I had disabled indicator lights leaving only perkey and underglow lights. As such, the numbers on the default code does not correlate to the actual light positions. Since indicator lights are bypassed in my case, light number 00 starts from the first underglow light. 

Secondly, I wasn't sure how the numbers carried over to the right(slave side} 35+?

So after some testing, I figured out that it was following  the LED chain during the build process. So rewrote all the LED positions into this neat little chart:
`// 10 11 20 21 30 31       66 65 56 55 46 45  
  //    00    01    02                 37    36    35
  // 09 12 19 22 29 32       67 64 57 54 47 44  
  //                
  // 08 13 18 23 28 33       68 63 58 53 48 43  
  //
  // 07 14 17 24 27 34       69 62 59 52 49 42  
  //
  //     06 15 16 25 26          61 60 51 50 41      
  //    05    04    03                 38    39    40`

These are the absolute positions of the LEDs (indicator disabled, only perkey and underglow enabled}. No guesswork, no math involved here. Just plug in the numbers and you're good to go. If you ONLY enabled perkey and bypassed the other lights, 06 on the chart above will be number 00 for you, and from there just rewrite the numbers, making sure you skip  the right side underglows.

 I tried plugging in the macros like so {L22, 6, hsv} but it wont compile so the above was the only way I could make it work. Perhaps there is an easier way to do this but this was already a handful for a beginner like me at this stage.

I hope this will help someone who might stumble upon this hurdle in the near future.

sanko, 2023-01-08:
Before I take a swing at it, did you ever get the matrix set up properly, @ryuugaplays? Having the underglow LEDs out of synch with the key LEDs during something like an RGB gradient is really bugging me.

ryuugaplays, 2023-01-09:
Not sure what you mean by "set up properly". As illustrated in my post above, the lights should be properly configured if you follow the diagram.

I did notice that in my testings, when using gradient or RGB wave lightings, the lights would flow from the opposite direction on the offhand (ie. RH-Left>Right  LH-Right>Left) 
Which is not the desired outcome of RGB gradient, but I suspect that has nothing to do with this topic. This topic is about telling QMK the correct LED coordinates according to it's physical circuit. RGB circuit is serial hence we have to follow the numberings. 

If you want to work on the gradient lighting, it's a separate topic than this one. One that has more to do about programming in QMK than circuitry. And no, I don't use gradients myself at the moment so I don't have that part of QMK set up. But I hope that helped.

closed 2023-01-09 by ryuugaplays