josefadamcik/SofleKeyboard #150: Help: Error while flashing QMK in Sofle Choc

Isaquelc, 2022-05-25:
**Kind of issue?**
_HELP!_

**Which Board?**
_Choc _  

**Which Board Revision?**
_Choc = 2.1_

**What steps have you taken**
* [ x] I have searched the Issues page
* [ x] I have read the documentation 
* [ x] For LED issues I have checked the following things 
        - the pro micro is programmed (and currently telling the LEDs to do something)
        - the pro micro is powered (and showing LEDs)
        - the LEDs are powered (check 5v across pins 1 and 3 of J1)
        - are any of the LEDs showing 

**What is the issue?**
While following the steps to flash QMK into the keyboard, I encounter the following error: 

 * The firmware size is approaching the maximum - 27892/28672 (97%, 780 bytes free)
Waiting for USB serial port - reset your controller now (Ctrl+C to cancel).....
Device /dev/ttyS5 has appeared; assuming it is the controller.
Remapped USB port to COM6
avrdude: ser_open(): can't open device "COM6": No such file or directory

avrdude done.  Thank you.

make[1]: *** [platforms/avr/flash.mk:175: flash] Error 1
Make finished with errors
make: *** [Makefile:413: sofle_choc:default:flash] Error 1

**What you expected:**
No errors

DaneEvans, 2022-05-25:
That's not a sofle issue. That's a qmk or computer issue.

I'd try a new cable, USB port and a different computer.

Failing that, search those issue strings on the qmk issues page

Isaquelc, 2022-05-25:
thank you, I'll check there and if I don't find anything, I'll create this issue there. I've tried different cables, ports and a different cmputer already, but it hasn't worked so far.

closed 2024-03-12 by DaneEvans