josefadamcik/SofleKeyboard #151: Gerber Version of Tenting Case for Sofle V2 and Sofle RGB

Smitty-Werbenjagermenjensen, 2022-06-01:
Laser cutting is sometimes not easy to find. It is also generally very very expensive, at least in my area. 

Could you create Gerber files for the top and bottom plates, so that we can print tenting case plates in FR-4 when ordering the PCB? 

I wonder if 1.6mm tenting case plates would be too thin or not. It would be very convenient to be able to buy the cases with the PCB.

josefadamcik, 2022-06-01:
Hi @Smitty-Werbenjagermenjensen , thanks for the suggestion!

There is one missing piece of information, you haven't mentioned which version of Sofle you are talking about. Is it for the RGB version?

From my side, I do not plan to do any adjustments to the tenting case and I would assume other maintainers might not be eager to do that as well. But contributions are always welcome! 

The tenting case (if I remember correctly) was submitted by a community member. I believe they might have been selling the case in their shop (sadly I do not remember any links or anything). So maybe you could try to google around, check Reddit etc. it might be possible to buy it somewhere.

> I wonder if 1.6mm tenting case plates would be too thin or not. I think it would be fine, especially as long as we put spacers between the two halves to fill the gap. It would be very convenient to be able to buy the tenting cases with the PCB.

People have most likely done that, but I personally have no experience with tenting. Maybe the best shot would be to test it.

DaneEvans, 2024-03-12:
The RGB has it already, and has had since release

closed 2024-03-12 by DaneEvans