josefadamcik/SofleKeyboard #158: Enhance: Kailh Choc v2 version?

rickyelopez, 2022-08-03:
Hey all,

Been checking out this keyboard for a while and I'm super tempted to build one. That being said, I think I would really like to use the Choc V2, which is not supported on any of these variants.

So, that led me to a few questions:
1. Is anybody working on an iteration of the Choc variant of this keyboard to support the v2 switches?
2. If not, is there a particular reason why not? Is there something wrong/bad about the v2 switches? As far as I can tell, they seem like a pretty clear improvement both in terms of feel, as well as support for keycaps (since they use the MX stem)
3. If the answers to the previous two question is "no", is there any interest in this? I am tempted to work on it, but would need something to actually hold myself to it :laughing:

brianlow, 2022-08-03:
No reason I'm aware of. 

Would be cool to try out. 

My impression is the v2 switches are less popular than others switches. Not all MX keycaps are compatible. 

The layout of all the variants, including Sofle Choc, use MX spacing. 

Might want a unique name. There will be confusion between the switch revision and pcb revision. e.g. the current Sofle Choc board is at version 2

rickyelopez, 2022-08-03:
I can't speak to their popularity to be honest, but glad to hear there isn't something wrong with them that I'm missing. Maybe I'll take a stab at adding that, then. Thanks!

tokazio, 2022-08-17:
I'm french, using azerty, it will be impossible to find nice rgb keycaps for choc v1, v2 should be great! How Can i help you to make one ?

rickyelopez, 2022-08-17:
Glad to hear someone else would enjoy it. Honestly I haven't taken a look at the board file yet to see how involved the changes would be, I will have to find some time to do that

JellyTitan, 2023-03-31:
I'm working on this! I started looking at Pimornoi integration and it sort of spiraled out of control. The choc v2's use the same hotswap sockets as V1 - but the third pin and bigger stem hole required a complete reroute.
While i was rerouting, I added a ton of tweaks.
I've got a prototype coming in the mail. It'll be at least one or two more rounds of prototypes before it's stable enough to open a PR, but you can take a peek if you're curious:
[image]

https://github.com/JellyTitan/SofleKeyboard/tree/choc_mx/Sofle_Choc_MX
https://github.com/JellyTitan/SofleKeyboard/blob/choc_mx/docs/build_guide_choc_mx.md

@brianlow I've been using the name "Sofle Choc MX" - is that sufficiently unique?

JellyTitan, 2023-04-05:
I'd be delighted to send you one of the prototypes when I get the bugs hammered out. (I sent out for another batch of boards today, the Choc V2 footprint was all loosey-goosey).   
Do you happen to know where i can find a writeup of enabling I2C on Sofle? 
I suspect that feature is not commonly used, since there's no QMK committed variants that enable it.
I'd like to pull it out if it's not popular.  Pulling it would let me swap in a smaller TRRS/TRS footprint, and I think the TRS option would get more use.

brianlow, 2023-04-07:
> I've been using the name "Sofle Choc MX" - is that sufficiently unique?

:+1: cool name, thanks for keeping the other boards in mind

> I suspect that feature is not commonly used

FWIW I don't think I've ever been asked about I2C on the Sofle Choc

JellyTitan, 2023-10-17:
Update for those interested in the Choc V2's:

I made 2 rounds of working prototypes, and then abandoned this project. 
I abandoned this for two reasons:
1. My design moves some case mounting holes, removes the oled jumpers, and tweaks the overall board outline. In retrospect this was a bad idea, because it makes this variant incompatible with other Sofle V2 cases. 
2. The ProMicro's aren't big enough for all the OLED/LED animations I'd like. (You can get fancier pro-micros that are bigger - but they are more expensive).

I don't have the gumption to go back and fix these things. 

If someone else want to try it out, or continue development, here's my branch. (The readme is half baked).
https://github.com/JellyTitan/SofleKeyboard/blob/choc_mx/docs/build_guide_choc_mx.md

While building this, I did get the choc/MX hybrid footprints dialed in well enough to be used in a [Cantor variant](https://github.com/JellyTitan/cantor-choc-V2), which is sadly stuck in [PR limbo](https://github.com/diepala/cantor/pull/25).