josefadamcik/SofleKeyboard #159: Help: I only get key presses from the plugged in side of the keyboard

dayorbyte, 2022-08-11:
**Kind of issue?**
_HELP!_

**Which Board?**
_RGB_  

**Which Board Revision?**
RGB = 2.1

**What steps have you taken**
* [X] I have searched the Issues page
* [X] I have read the documentation 
* [X] I reset the eeprom
* [X] I installed the default firmware instead of my custom firmware
* [X] I googled a lot
* [X] I tried a different TRRS cable

**What is the issue?**

My keyboard was working perfectly for the last few months since I got it (purchased pre-soldered), but today I plugged it in and the keys on the plugged in side work but not the other side. If I swap the USB-C cable then just that side works. The RGBs are working on both sides, so power is flowing, but seemingly no data.

**What you expected:**

The side of the keyboard attached by the TRRS cable can transmit keys

dayorbyte, 2022-08-11:
The Etsy seller I got the keyboard from suggested it could be related to power draw, and when I turned down the brightness on each half and replugged it it worked. The default firmware didn't fix it since it also starts at full brightness. Interestingly, once the halves are working I can turn up the brightness and it keeps working.

It's easy enough to modify my firmware to work around this, but since it had worked for several months I was wondering if this is an issue that's come up with extended use before and if it might be something degrading over time.

DaneEvans, 2022-08-12:
More likely the USB port has more load, and supplied a lower voltage
today.

Or the USB cable has been damaged increasing the voltage drop to the
keyboard.

Or the TRRS cable has been damaged slightly, increasing the voltage drop
across it.

Degradation of the Arduino boards themselves is unlikely in room
temperature, and would be unlikely to be symmetrical.

dayorbyte, 2022-08-12:
That makes sense. I'm glad to hear that it's unlikely to be the keyboard hardware itself. Closing the bug since I don't think there's anything else to be done. Thanks again!

closed 2022-08-12 by dayorbyte