josefadamcik/SofleKeyboard #161: Help: Are these contacts from the V1.1/2.1 equivalent to these contacts on the V2?

Moghul, 2022-08-15:
**Kind of issue?**
_HELP!_

**Which Board?**
Basic

**Which Board Revision?**
2.0

**What steps have you taken**
* [x] I have searched the Issues page
* [ ] I have read the documentation (I don't know how to read the pcbs and even if I found the exact information I'd check with you)

**What is the issue?**
I want to add a Pimoroni Trackball Breakout to my Sofle V2. I've found instructions for how to do it on a sofle 1.1 and 2.1 [here](https://github.com/foureight84/sofle-keyboard-pimoroni). 

According to those docs, I need to solder the Pimoroni to these contacts:
[image]

However, on the Sofle V2 they've been moved. Are these contacts from the Sofle V2 the same thing? Can I solder the Pimoroni to these? Please excuse the scuffed screenshots, I ripped them out of a reddit post.
[image]

[image]

**What you expected:**
N/A, this is some custom stuff and the keyboard is awesome anyway

DaneEvans, 2022-08-15:
Yeah, those will work to get access to the I2C lines.

I'd go off the through holes where the Arduino mounts, you have less chance
of overheating the copper and having the pads peel up.

If you decide not to run the OLED, you could go off the 2.54 header there,
but you need to get it around the right way.
You could also do it and keep the OLED, but it may be more ugly

Moghul, 2022-08-16:
Thank you so much for the reply! I'll be taking your advice and soldering to the chip's pins directly.
I do want the OLED and I've already soldered it on, so I think I won't be messing with that.

closed 2022-08-16 by Moghul

etrnal70, 2023-04-23:
@Moghul hello, sorry for reviving this rather old issue. I'm also trying to add a pimoroni trackball to my v2 pcb. I want to ask which pin you use to get it working, the middle one or the side one ? A pics of your wiring would be greatly appreciated. Thanks ! 😄