josefadamcik/SofleKeyboard #163: One column and one row suddenly not working

F1orian, 2022-10-10:
Ijust fineshed building the RGB Version and tested all keys successfully before soldering the LEDs. Now I've soldered all the LEDs and everything seemed to work, but upon testing all the keys again, I was greeted with one dead row and one dead column. LEDs all still work.

[image]

What could be the cause of this, and how do I troubleshoot this?
Thanks a lot in advance.

DaneEvans, 2022-10-11:
Check that the switches are all plugged in properly.

Then look at the switch and diodes around the common key for both rows.
It's probably a bad diode. I think a shorting diode presents that way, I'd
need to check the schematic to be sure.

If you have them spare, it's probably easier to just replace both of them.

F1orian, 2022-10-13:
Thanks for taking a look!

I've checked the schematic but I was unable to identify the diode(s) that I should replace. Also I'm a bit skeptical because they all worked before. Are they sensitive to heat, like the LEDs?

Looking at the schematic, I don't think there is a single diode that affects all switches of col2:
[image]

There are however a few diodes in row0:
[image]

Sadly I have no experience at all when it comes to reading schematics, so if you could take another look at this, before I start de-soldering and potentially breaking something for real, I would really appreciate that!

DaneEvans, 2022-10-13:
> I'll have a look, but for starters, that's the layout.  What we need to
> look at is the schematics.
>
>
>
> The thing to know with these is that it 'scans'
>
>
> First we start with one axis - a single column Gets pulled high, and then
> we poll each of the rows, and see if they are high If so the common switch
> is pressed.
>
> So, column 2 being all dead - no rows get pulled high, all floating
> - so a fw issue (unlikely if it's not symmetrical)
> - the column trace never goes high - check continuity to the micro
> - the trace is getting pulled down more heavily to gnd than it can provide
> - check no continuity to gnd.
>
> Row 0 all dead - this trace never gets pulled up.
> - Could be a short to gnd. - use a multimeter
> - could be disconnected from the micro.
> - could be all 4 switches on the trace.  (Pretty unlikely)
>
> The most likely single cause would be a short to gnd around the common
> switch or its diode.  Anything else means that you have at least 2 issues.
>
> If that doesn't get you anywhere, I'd start reading up on the purpose of
> the diodes and trying to understand the matrix, because while I have a few
> other things I'd try to narrow it down, you won't get much out of them if
> you don't understand the circuit.
>
>
> Also what board is that that implemented length matching? That seems
> pretty unnecessary. For out purposes.  It does look cool though.

One other thing I'll note.  The highlighted row and column on your post are
adjacent pins at the micro end.  If you've been heavy handed with the
solder there could be a short there too.

If the columns get driven low when they aren't in use, then a simple short
between those, or of the common diode would probably cause all your
issues.  I can't recall if qmk does that, but I have a feeling it does.

Also looks like that screen shot is my board.  Why the hell did I length
match those traces?

F1orian, 2022-10-13:
Alright, thanks a lot for your help so far! 
I'll get a multi-meter on the weekend and then I'll check the things you suggested.

I just did some cleaning-up on the pins at the micro on the PCB-side (quite sure there's nothing shorted on this side), and the pins for col2 and row0 on the micro itself. After that, col2 was working for around a minute, but weirdly it stopped working again. Row0 remained unchanged.

DaneEvans, 2022-10-13:
In that case I'd say 2 problems.
One is a loose pin or dry solder joint at the microcontroller on that
column.  It worked for a while after you made it connect as you poked at
it.

If you're using sockets, that leg may be too short.

But yeah, a metre is pretty much the minimum trouble shooting gear you
need.

F1orian, 2022-10-17:
So I got the metre and did some testing: 

**Row0**
I have continuity to gnd on all switches (including row0) **except for col2**
Does that mean that row0 is just not connected to the micro correctly but everything else in the chain is soldered correctly and working?

**Col2**
I have continuity on col2 to the pin that connects to the micro for all switches of col2.

I am not quite sure how to proceed with these findings...

What options does that leave me with, also considering that col2 was briefly working for around a minute?

closed 2024-03-12 by DaneEvans