josefadamcik/SofleKeyboard #166: Help: Some per-key LEDs of the slave-side stuck at one color/brightness

lauchlan105, 2022-11-12:
## **Kind of issue?**
_HELP!_

## **Which Board?**
RGB (not keyhive)

## **Which Board Revision?**
v2.1 (Dane Evans)

## **What steps have you taken**
* [x] I have searched the Issues page
* [x] I have read the documentation 
* [x] For LED issues I have checked the following things 
        - the pro micro is programmed (and currently telling the LEDs to do something)
        - the pro micro is powered (and showing LEDs)
        - the LEDs are powered (check 5v across pins 1 and 3 of J1)
        - are any of the LEDs showing 

## **LED configuration?**
Indicator + drop lighting + per key lighting

## **What is the issue?**
I've finished building both halves.

If I plug in just the left half, all LEDs work (color & brightness).
If I plug in just the right half, all LEDs work (color & brightness).

However, if I connect the two halves, LEDs 35 & 36 on the **_opposite_** side are stuck at different colors, and full brightness. I say opposite side, because if I plug the USB into the right half, LEDs 35 & 36 on the left half then have this problem.

Plugged in LHS
[image]

Plugged in RHS
[image]

## **Notes**
- **Firmare**: I have tried sofle_v1, sofle_keyhive, and both of their _Via_ equivalents with no luck
- I have (painstakingly) removed the micro controller twice from each side, so it isn't impossible the PCB has suffered damage (the micro controllers are not damaged though as I was replacing them)
- I am using Elite-C V4 controllers
- I _did_ have to do a janky bridge (see images) as one of the traces on the top half of the RHS board lifted when I was trying to remove the old pins. The symptoms of this were...

Fix I had to do for the RHS as one of the traces for the controllers lifted, and keys 13-18 weren't working until this. Although I found this after the fact, the images in [this reply](https://github.com/josefadamcik/SofleKeyboard/issues/163#issuecomment-1277367148) might help make sense of what this bridge is supposed to do.
[image]

## **What you expected:**
- LEDs 35 * 36 follow the colors/patterns of all other per-key LEDs

DaneEvans, 2022-11-12:
Does it still happen at low brightness/single colour?

lauchlan105, 2022-11-15:
> Does it still happen at low brightness/single colour?

Yeah - any colour or brightness, these lights stay *mostly* the same. The purple one seems to get more pink, or more purple when the defined colour is close to that colour.

lauchlan105, 2022-11-21:
Turns out the firmware I had installed had the LED count set to `70` instead of `72`

closed 2022-11-21 by lauchlan105