josefadamcik/SofleKeyboard #168: Help: short somewhere, need schematic to trace

rob31183, 2022-11-26:
HELP!_

RGB

RGB =  2.1

**What steps have you taken**
* [Yes ] I have searched the Issues page
* [Yes ] I have read the documentation 
* [Yes ] For LED issues I have checked the following things 
        - the pro micro is programmed (and currently telling the LEDs to do something)
        - the pro micro is powered (and showing LEDs)
        - the LEDs are powered (check 5v across pins 1 and 3 of J1)
        - are any of the LEDs showing 

**What is the issue?**
Backlight led on right hand will not work (left works fine).  1st led works, but 2 of the 4 led contacts are grounded so I wasn't getting signal to led 2.  I assume I messed it up somewhere upstream.

I tried to install kicad to get the schematics so could try and trace the issue, but I can't get the .pro and .sch files to work.

**What you expected:**

Pdf of the schematic uploaded to the git would be nice
Any advice on how I could have grounded one of the data pads

DaneEvans, 2022-11-28:
What do you mean you can't get them to work? Are the symbols not appearing?
Do you need to link the Library that is in the repo?

It's kicad, the thing is awful. You'd think that linking would be in the
project, but apparently not.

A pdf would definitely be useful, I'm surprised that there isn't one in
there.  Or in the documentation section.   I don't currently have an
environment, so I can't make one at the moment.

The most likely would be at the last LED to work, or the next one.  Sounds
like you've grounded the data trace.  Or possibly cooked an LED.

rob31183, 2022-12-04:
I was able to resolve the grounding issue, and the board no longer shorts
out.  It was some sort of issue with the jumpers.  However, the per key
LEDs still won't turn on.

I replaced the last led on the underglow Incase it wasn't sending data, and
the first (only) led on the per key, and it still won't turn on.

I also managed to rip up one of the traces, so I had to jump + from another
led to that one.

I can see I have 4.42 volts going across the led.  I traced data out of the
last underglow led to data in of the first per key led, and I can trace
data out to the second per key of the led.

I even swapped my pro micros between sides, and it did not resolve the
issue.

I don't know where to go from here.  What else could it be that the first
per key led won't turn on, but only on the right side?

DaneEvans, 2022-12-04:
Are you sure that the firmware is set to drive enough LEDs? Are you sure
that it's asking for the latter ones to be turned on.

If it's not power, then it's data, or the LED itself.
Measure data at the back of the LED if you can, try jumping off that to a
strip if you have one.

rob31183, 2022-12-05:
I tried to improvise a strip, and it might not have worked.  I got a tiny
green light when nothing was tied into din.  When I touched din to
anything, the light turned out.

If I put a multi on ground, and any other din or dout i get .05mv.  When I
touch it to the dout of the last underflow led i get .05mv.  when I touch
it to din for the first per key led, I get .01mv.  it sounds like a wire
issue between the LEDs, but I already tried jumping it out.  Could it be a
jumper (j4/5/6) issue?  I'll redo them again I guess....

I pulled the firmware from this site, since that is where I bought the kit
from.  I haven't tried compiling a fresh .hex from the git hub site, I'm
not really sure how to do that yet.  I'll try that tomorrow.

https://docs.beekeeb.com/build-guide/sofle-rgb-v2.1-soflekeyboard-build-log-guide-with-photos

rob31183, 2023-01-20:
Hi.  I just realized that I never responded to this.

I got it all working, I have no clue how.

Thank you very much for your help, I've attached a pretty picture.

I still need to figure out programing to make those screens useful.

After using it for a while, I love it.  My only request for a future
version would be another column on the outside of each half for a few more
keys.

closed 2024-03-12 by DaneEvans