josefadamcik/SofleKeyboard #172: A link in Sofle Choc build doc is not available

Ross-Li, 2023-02-01:
Link in this doc https://github.com/josefadamcik/SofleKeyboard/blob/master/docs/build_guide_choc.md

at line:
+ Make sure your QMK environment [is setup](https://beta.docs.qmk.fm/newbs/newbs_getting_started).

is not available. Here is the error message when opened:
[image]

Possible updated link of qmk "getting started" guide is:
https://docs.qmk.fm/#/newbs_building_firmware_configurator

(Sorry to submit this in the software bug issue section, as I can't find a documentation issue issue section.)

JellyTitan, 2023-03-21:
Submitted PR to fix: https://github.com/josefadamcik/SofleKeyboard/pull/176

DaneEvans, 2024-03-12:
Closed, as it got merged

closed 2024-03-12 by DaneEvans