josefadamcik/SofleKeyboard #173: LED 36 doesn't light up when using `rgb_default` keymap on either side

benjamin051000, 2023-02-10:
**Kind of issue?**
Software/HELP!

**Which Board?**
RGB

**Which Board Revision?**
RGB 2.1

**What steps have you taken**
* [x] I have searched the Issues page
* [x] I have read the documentation 
* [x] For LED issues I have checked the following things 
        - the pro micro is programmed (and currently telling the LEDs to do something) Yes, via `qmk flash --keyboard sofle/rev1 --keymap rgb_default  --bootloader dfu` 
        - the pro micro is powered (and showing LEDs) Yup
        - the LEDs are powered (check 5v across pins 1 and 3 of J1) Just checked. Yes
        - are any of the LEDs showing? All but 36

**What is the issue?**
_Please also include what other components you are having issues with - is the software QMK or the configurator, include pictures of your LEDs or soldering issues etc._

 
[image]
You can see that LED 36 is not illuminated (observe lack of reflection through the LED hole).

**What you expected:**

* [ ] LED 36 should light up just like the others

At first I thought I just botched the left half's LED 36... Every single LED works reliably except 36. I replaced 36 several times and considered replacing 35 (perhaps 35 dout was broken). But I just finished the right half, and the exact same issue arose. I'm wondering if this could potentially be a firmware bug since it's unlikely that I made _identical_ mistakes twice in a row, especially considering the chance that it would have happened on any other LED.

DaneEvans, 2023-02-10:
Have a search for closed LED bugs here.  I think it's a known issue.

I think the default is without the top facing indicator. I thought someone
had fixed it in qmk, but the recent reshuffle may have seen it reappear, or
it may never have been fixed there.

qjao, 2024-01-05:
[This page ](https://docs.beekeeb.com/build-guide/sofle-rgb-v2.1-soflekeyboard-build-log-guide-with-photos) has a firmware with it working.

7oo1er, 2024-02-24:
Hello.

I use indicator LED too and changed in config.h where it appears 35 to 36:

```
#define RGBLED_NUM 36
#define RGBLED_SPLIT { 36, 36 }
```

closed 2024-03-12 by DaneEvans