josefadamcik/SofleKeyboard #174: Sofle v2 layout that is compatible with Kailh Choc v1 switches

jwrussert, 2023-02-10:
I love the pinky offset of the Sofle v2, but I want to use Kailh Choc v1 switches. From what I can see, I would have two options:

1. Sofle_v2, which doesn't support these switches
2. Sofle_Choc, which doesn't have the full v2 pinky stagger.

I'm happy to do this myself, and make a PR. I just don't want to spend the time if it's already done or in progress.

JellyTitan, 2023-10-17:
I built this! I made 2 rounds of working prototypes, and then abandoned the project. 
I abandoned this for two reasons:
1. The ProMicro's aren't big enough for all the OLED/LED animations I'd like. (You can get fancier pro-micros that are bigger - but they are more expensive).
2.  My design moves some case mounting holes, removes the oled jumpers, and tweaks the overall board outline. In retrospect this was a bad idea, because it makes this variant incompatible with other Sofle V2 cases. 
I don't have the gumption to go back and fix these things. 

If you want to try it out, or continue development, here's my branch. (The readme is half baked).
https://github.com/JellyTitan/SofleKeyboard/blob/choc_mx/docs/build_guide_choc_mx.md

I did use the choc/MX hybrid footprints in this project to build a [Cantor variant](https://github.com/JellyTitan/cantor-choc-V2), which is sadly stuck in [PR limbo](https://github.com/diepala/cantor/pull/25).

razorree, 2025-04-20:
do you use decreased spacing on PCB? (less than MX)