josefadamcik/SofleKeyboard #175: I can't seem to get the Serial communication between the two halves working

jeremiah-miller, 2023-02-12:
**Kind of issue?**
Assembly 

**Which Board?**
Original

**Which Board Revision?**
1.1

**What steps have you taken**
* [x] I have searched the Issues page
* [x] I have read the documentation 
* [x] For LED issues I have checked the following things 
        - the pro micro is programmed (and currently telling the LEDs to do something)
        - the pro micro is powered (and showing LEDs)
        - the LEDs are powered (check 5v across pins 1 and 3 of J1)
        - are any of the LEDs showing 

**What is the issue?**
_Please also include what other components you are having issues with - is the software QMK or the configurator, include pictures of your LEDs or soldering issues etc._
I have my keyboard all soldered and everything, however I'm having issues getting the two halves to communicate. Here are the relevant things in the config.h files:

```
#define USE_SERIAL
#define EE_HANDS 
#define FORCED_SYNC_THROTTLE_MS 100
#define SPLIT_CONNECTION_CHECK_TIMEOUT 500
#define SPLIT_TRANSPORT_MIRROR
#define SPLIT_MODS_ENABLE
#define SERIAL_DEBUG
#define SELECT_SOFT_SERIAL_SPEED 1
#define SOFT_SERIAL_PIN D2
```

I flash with `qmk flash -kb sofle/rev1 -km foureight84 -bl dfu-split-right` for the right side and `qmk flash -kb sofle/rev1 -km foureight84 -bl dfu-split-left` for the left, and whichever side is connected to the computer works as expected.

I've also tried wiring vcc->vcc, gnd->gnd, and D2->D2 directly, instead of using the trrs cable, but not even that worked. The other half gets power, which I know since all the leds turn on, but keys on the secondary half don't work

**What you expected:**
* [ ] keys on both halve of the keyboard to work when using a trrs cable or wiring the necessary serial pins directly

DaneEvans, 2023-02-13:
Check your connectivity all the way through, micro to micro.

But that sounds like a firmware issue. Take it to the qmk GitHub.

jeremiah-miller, 2023-02-13:
I did wire the two halves directly together to not really on the trrs, but sounds good, i'll ask there

DaneEvans, 2023-02-13:
That's still missing the link between the Arduino and the Sofle board.

jeremiah-miller, 2023-02-13:
What do you mean? Both halves work as expected independently, it's only when connecting them that the slave side doesn't work. It still gets power since the leds turn on, just no data is transferred

DaneEvans, 2023-02-13:
If that serial line is broken at any point, you'll get those symptoms.
That includes:
- micro to Arduino board
- Arduino to pins
- pins to sockets
- sockets to Sofle
- Sofle to trrs connectors
- trrs connector to cable
- trrs cable.

You've eliminated the last 3 by direct wiring.  The first is very unlikely,
but I'd throw a new Arduino on in both combinations and see if that fixes
it.

If you measure from the Arduino to Arduino on that data line then you can
eliminate the rest. Which will just leave FW

DaneEvans, 2024-03-12:
Closing as linked QMK issue is closed

closed 2024-03-12 by DaneEvans