josefadamcik/SofleKeyboard #177: Help: Sofle RGB v3 Basic Questions - led and micro positioning

DimitarStaykov, 2023-04-08:
**Kind of issue?**
Assembly

**Which Board?**
 RGB 

**Which Board Revision?**
RGB = v3

**What steps have you taken**
* [x ] I have searched the Issues page
* [x ] I have read the documentation 

**What is the issue?**
Should be some pretty basic questions.

[image]

1. Most guides mention a triangle that shows which pad is ground for the led. On the v3 board there are no triangles, but 1 of 4 pads is surrounded by a white line. Is that the designated ground? The ones kind of surrounded by red on the picture.
2. In green - There are two sets of slots for the pro micro. The guide says for v1(both controller's chips are facing the board) use the sockets marked by the rectangle. And I am not 100% sure so I need to ask. For v1 do we use the sockets that are inside a rectangle or the ones that are on one of his sides.

Thanks.

DaneEvans, 2023-04-09:
Read the documentation

MohamadSaada, 2023-12-13:
Yes I have to agree that V3 led positioning support is non existing, there is no documentation that discuss the white rectangles. I followed the one on Keebs as it's the one that with most resemblance to the V3 board and that didn't work. So I am stuck with a Sofle that doesn't light up which seems the norm amongst many Sofle users, as most get frustrated at this point and then give up on the rgb lighting which is a shame.

DaneEvans, 2023-12-13:
Really not sure how to make it any clearer that you need to read the documentation. 
[image]
https://josefadamcik.github.io/SofleKeyboard/build_guide_rgb.html

closed 2023-12-13 by DaneEvans

MohamadSaada, 2023-12-13:
This guide doesn't indicate that the white rectangles around one of the pads on v3 is the same as the white dot in this guide. I've followed the guide but it still didn't work. I am using KB2040 and maybe that's the problem. Thank you for the quick reply and help. Sorry to bother.

DaneEvans, 2023-12-13:
The white marking is always on pin 1. And a cursory glance at the PCB files
or schematic will confirm that.

The problem is that there are at least two varieties of LED, with different
pin 1 markings.  Making any form of distinction kind of pointless. All the
info is in the guide, this is a hobbyist board, several years since any of
the maintainers last made one