josefadamcik/SofleKeyboard #183: Question: Difference between v1 and v2.

HasbersHz, 2023-08-12:
It's very funny. ... Long story short, at first I thought it was the first version, then I looked at the layout of the keys and saw that it was the second version. So I'll probably order it.
https://aliexpress.ru/item/1005005352486359.html?sku_id=12000032715330332
It was cheaper to assemble myself, but I don't really want to do it. And I would also like to point out that not all links in the guide work. Thank you for keyboard.
I know that something like this should not go to the issues, but rather to the mail, but I did not find it.
And finally, the question, do the versions really differ so much? I was very interested in spending money on something, which of course is pointless, but I can still do that, or I'll give up on this idea. And I haven't used separate keyboards, so I'm wondering if the difference between the keyboards is only in the offset?

JellyTitan, 2023-10-17:
If you find dead links, please submit a pull request to correct them. Alternately, if you can tell me which links are dead, i'd be happy to setup a pr for those.

Between V1 & V2, it's mostly tweaking the layout and component placement. If this is your first split keyboard, either version should be suitable. 
You can find a brief comparison of the differences [here](https://josefadamcik.github.io/SofleKeyboard/), and an in depth write up of the projects evolution [here](https://josef-adamcik.cz/electronics/soflekeyboard-evolving.html).

HasbersHz, 2023-10-17:
> If you find dead links, please submit a pull request to correct them. Alternately, if you can tell me which links are dead, i'd be happy to setup a pr for those.
>
> Between V1 & V2, it's mostly [...]

Thank`s, but i already using Sofle for a month or so.