josefadamcik/SofleKeyboard #184: Enhance: Sofle Choc: Automatic left/right detection

t-8ch, 2023-09-06:
**What feature would you like to see?**

Currently the left/right detection in the Sofle Choc is done by assuming the cable is plugged into the left half.
(As proposed in the code at https://github.com/qmk/qmk_firmware/pull/16736 )
This restricts the user a bit.

QMK can also detect the sides based on a GPIO pin.
As the pin `D4` of the controller is not used on the current PCB there could be a jumper or direct connection from D4 to VCC on one and GND on the other PCB side.

**Are you prepared to help?** 
My EDA skills are limited and I already have my Sofle Chocs and can work around it with a physical wire :-)

Cc @brianlow

t-8ch, 2023-10-04:
So a jumper doesn't really make sense.

Instead of connecting both `D4` holes between the two mounting sides of the pro-micro should not be connected with each other instead one is connected to VCC and one to GND.

7oo1er, 2024-02-24:
Hello.

I use RGB version with a RP2040. Config.h looks like:

```
// #define MASTER_LEFT
// #define MASTER_RIGHT
#define EE_HANDS

#define SPLIT_USB_DETECT
#define SPLIT_USB_TIMEOUT_POLL 10
#define SPLIT_WATCHDOG_ENABLE
#define SPLIT_WATCHDOG_TIMEOUT 3000
```

Builds with:
`qmk flash -c -kb sofle -km rgb_default -e CONVERT_TO=promicro_rp2040 -bl uf2-split-left`
`uf2-split-left` or `uf2-split-right` for each side.

Not sure if it`s fully correct, but in this case USB can be plugged in any side with no mirror effects.

t-8ch, 2024-02-24:
>  uf2-split-left or uf2-split-right for each side

This is what I would like to avoid personally.
It was easier to solder a cable once than having to remember to flash the correct image every time.

7oo1er, 2024-02-24:
Got it.

I removed `-bl uf2-split-...` key and flashed both sides with a same file. Both sides are _right_ side now, can type 7890 on both with any side USB connected. Funny. And LED colors are same/mirrored.

I had errors with right encoder change volume, with a PGDN same time, was searching errors with (maybe) wrong encoder side recognition.