josefadamcik/SofleKeyboard #185: Can I use the bottom plate from sofle v2 on sofle choc

jittapont, 2023-09-08:
**Kind of issue?**
_Bug / Assembly / Software / HELP!_
I need some help. Can I use the bottom plate from sofle v2 (https://github.com/josefadamcik/SofleKeyboard/releases/download/v2.0/v2-gerber-bottom.zip) on sofle choc (https://github.com/josefadamcik/SofleKeyboard/tree/master/Gerbers/Choc)

**Which Board?**
_Original / RGB / Choc _  
_Please note that Keyhive is selling a customised and incompatible Sofle RGB version which is not part of this repository and therefore not supported here._

**Which Board Revision?**
_This is the one written on main pcb that you are using. 
On the original it is written on the top outside corner on the silk screen. 
On the RGB it is written on the bottom outside corner, in copper under the soldermask_
Original = 1.0, 2.0 
RGB =  2.0, 2.1
Choc = 2.1

**What steps have you taken**
* [ ] I have searched the Issues page
* [ ] I have read the documentation 
* [ ] For LED issues I have checked the following things 
        - the pro micro is programmed (and currently telling the LEDs to do something)
        - the pro micro is powered (and showing LEDs)
        - the LEDs are powered (check 5v across pins 1 and 3 of J1)
        - are any of the LEDs showing 

**What is the issue?**
_Please also include what other components you are having issues with - is the software QMK or the configurator, include pictures of your LEDs or soldering issues etc._

**What you expected:**

* [ ] item 1
* [ ] item 2

brianlow, 2023-09-08:
No. The Sofle RGB is closer: same outline but mounting holes are in different locations due to routing needs. It wouldn't be hard to create one. Clone the top plate and delete the switch holes.

jittapont, 2023-09-08:
Can I clone the PCB and then remove all the components instead of the top plate?

brianlow, 2023-09-08:
Yeah should be able to. One way to check your work is export to SVG and make sure sure the outlines and mounting holes line up.