josefadamcik/SofleKeyboard #187: SW: Firmware size is too large after adding the mouse key support

jittapont, 2023-10-11:
**Which Board?**
- Sofle Choc

**Which Board Revision?**
Choc = 2.1

**What Software are you using?**
- qmk

**What is the issue?**
_Please also include what other components you are having issues with -include any code snippets, error messages or the like that may help. Include pictures of your LEDs or soldering issues etc._
After adding the `MOUSEKEY_ENABLE = yes` in `rules.mk` and compiled the firmware. I got an error about the firmware size is too large.
[image]

QMK version
[image]

**What you expected:**
- Can compile the firmware with no errors

jittapont, 2023-10-11:
I had disabled some features according to this docs but got the same error 😢

brianlow, 2023-10-12:
Maybe remove more features?

Here is post on reducing firmware size, with size of various features:

https://thomasbaart.nl/2018/12/01/reducing-firmware-size-in-qmk/#memory-cost-per-feature