josefadamcik/SofleKeyboard #194: Add VIA support for Sofle Choc

JellyTitan, 2024-01-11:
I'd like to add VIA support for the Sofle Choc.
Since there's already a via folder on the QMK side, the config on the VIA side needs to be added.
I didn't see any open PR's, so I went ahead and submitted one for the Sofle Choc: https://github.com/the-via/keyboards/pull/2139
All PR's are paused on VIA for December/January, so it'll sit for a bit.

JellyTitan, 2024-02-14:
The VIA PR is [blocked ](https://github.com/the-via/keyboards/pull/2139#discussion_r1477167695) because in the QMK branch the Product ID and the VendorID are the savme as the Sofle rev1.
[image]

brianlow, 2024-02-14:
@JellyTitan go for it!

JellyTitan, 2024-07-01:
The PR for the vid/pid choc was closed and moved to a new PR. (Same code).
https://github.com/qmk/qmk_firmware/pull/24030

JellyTitan, 2024-07-04:
@brianlow The PR to make the [Sofle Choc VID/PID unique](https://github.com/qmk/qmk_firmware/pull/24030) has been approved and merged into QMK master. The VIA PR has been updated.

brianlow, 2024-07-05:
Great, let me know if you need anything

JellyTitan, 2024-07-07:
VIA support has been [approved and merged](https://github.com/qmk/qmk_firmware/pull/24030). @brianlow is documentation support warranted?

brianlow, 2024-07-08:
Up to you

closed 2024-07-08 by JellyTitan