josefadamcik/SofleKeyboard #196: SW: wrong LED config numbers, indicator and glow issues

7oo1er, 2024-02-25:
**Which Board?**
_RGB_

**Which Board Revision?**
_RGB = 2.1_
Build with indicator light, backlight and per key. RP2040 as PM replacement

**What Software are you using?**
_QMK, rgb_default_

**What is the issue?**

**1. Wrong indicator brightness, wrong LED number:**

Change `keymap.c`
```
// Light combinations
#define SET_INDICATORS(hsv) \
    {0, 1, HSV_OVERRIDE_HELP(hsv, INDICATOR_BRIGHTNESS)}, \
    {35+0, 1, hsv}
```
To
```
// Light combinations
#define SET_INDICATORS(hsv) \
    {0, 1, HSV_OVERRIDE_HELP(hsv, INDICATOR_BRIGHTNESS)}, \
    {35+1, 1, HSV_OVERRIDE_HELP(hsv, INDICATOR_BRIGHTNESS)}
```

**2. Some keys LEDs lighten like solid backlight, to remove it and fix indicator number:**

Change `keymap.c`
```
#define SET_LAYER_ID(hsv) 	\
	{0, 1, HSV_OVERRIDE_HELP(hsv, INDICATOR_BRIGHTNESS)}, \
    {35+0, 1, HSV_OVERRIDE_HELP(hsv, INDICATOR_BRIGHTNESS)}, \
		{1, 6, hsv}, \
    {35+1, 6, hsv}, \
		{7, 4, hsv}, \
	  {35+ 7, 4, hsv}, \
		{25, 2, hsv}, \
	  {35+ 25, 2, hsv}
```
To
```
#define SET_LAYER_ID(hsv) 	\
    {0, 1, HSV_OVERRIDE_HELP(hsv, INDICATOR_BRIGHTNESS)}, \
    {35+1, 1, HSV_OVERRIDE_HELP(hsv, INDICATOR_BRIGHTNESS)}
```

**3. Fix total LED number. LED effects goes to keyboard center. To make it go same direction on both sides (left to right) and make backlight glow with near keys in same color:**

Change `config.h`
```
#ifdef RGB_MATRIX_ENABLE
#define RGBLED_NUM 35    // Number of LEDs
#define RGBLED_NUM 35    // Number of LEDs
#define RGB_MATRIX_LED_COUNT RGBLED_NUM
#endif
```
To
```
#ifdef RGB_MATRIX_ENABLE
#define RGBLED_NUM 36
#define RGB_MATRIX_LED_COUNT RGBLED_NUM
#endif
```

Change `config.h`
```
#ifdef RGBLIGHT_ENABLE
    #undef RGBLED_NUM

	//#define RGBLIGHT_EFFECT_BREATHING
	#define RGBLIGHT_EFFECT_RAINBOW_MOOD
	//#define RGBLIGHT_EFFECT_RAINBOW_SWIRL
	//#define RGBLIGHT_EFFECT_SNAKE
	//#define RGBLIGHT_EFFECT_KNIGHT
	//#define RGBLIGHT_EFFECT_CHRISTMAS
	//#define RGBLIGHT_EFFECT_STATIC_GRADIENT
	//#define RGBLIGHT_EFFECT_RGB_TEST
	//#define RGBLIGHT_EFFECT_ALTERNATING
	//#define RGBLIGHT_EFFECT_TWINKLE

    #define RGBLED_NUM 70
	//#define RGBLED_SPLIT
	#define RGBLED_SPLIT { 35, 35 } // haven't figured out how to use this yet

	//#define RGBLED_NUM 30
    #define RGBLIGHT_LIMIT_VAL 120
    #define RGBLIGHT_HUE_STEP 10
    #define RGBLIGHT_SAT_STEP 17
    #define RGBLIGHT_VAL_STEP 17
#endif
```
To
```
#ifdef RGBLIGHT_ENABLE
    #undef RGBLED_NUM

	//#define RGBLIGHT_EFFECT_BREATHING
	#define RGBLIGHT_EFFECT_RAINBOW_MOOD
	//#define RGBLIGHT_EFFECT_RAINBOW_SWIRL
	//#define RGBLIGHT_EFFECT_SNAKE
	//#define RGBLIGHT_EFFECT_KNIGHT
	//#define RGBLIGHT_EFFECT_CHRISTMAS
	//#define RGBLIGHT_EFFECT_STATIC_GRADIENT
	//#define RGBLIGHT_EFFECT_RGB_TEST
	//#define RGBLIGHT_EFFECT_ALTERNATING
	//#define RGBLIGHT_EFFECT_TWINKLE

    #define RGBLED_NUM 72

    // this to flip right side effects, make same direction on both sides. 36 = indicator LED. But underglow looks very different.
    // #define RGBLIGHT_LED_MAP { 0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31, 32, 33, 34, 35, 36, 71, 70, 69, 68, 67, 66, 65, 64, 63, 62, 61, 60, 59, 58, 57, 56, 55, 54, 53, 52, 51, 50, 49, 48, 47, 46, 45, 44, 43, 42, 41, 40, 39, 38, 37 }

    // use key columns and backlight LEDs. Cool solid color on both sides. Use it.
    #define RGBLIGHT_LED_MAP { 0, 6, 17, 29, 34, 22, 10, 11, 1, 2, 3, 4, 5, 7, 8, 9, 21, 23, 12, 13, 14, 15, 16, 18, 19, 20, 33, 35, 24, 25, 26, 27, 28, 30, 31, 32, 36, 66, 55, 43, 38, 50, 62, 61, 71, 70, 69, 68, 67, 65, 64, 63, 51, 49, 60, 59, 58, 57, 56, 54, 53, 52, 39, 37, 48, 47, 46, 45, 44, 42, 41, 40 }

	//#define RGBLED_SPLIT
	#define RGBLED_SPLIT { 36, 36 } // haven't figured out how to use this yet //tooler

	//#define RGBLED_NUM 30
    #define RGBLIGHT_LIMIT_VAL 120
    #define RGBLIGHT_HUE_STEP 10
    #define RGBLIGHT_SAT_STEP 17
    #define RGBLIGHT_VAL_STEP 17
#endif
```

**What you expected:**

* [ ] no error colors

[image]

JellyTitan, 2024-03-06:
@DaneEvans Can you take a gander?

7oo1er, 2024-03-08:
4. Right rotary give keys and changes volume also. Fix to remove right volume change:

Change `keymap.c`, return false.

```
bool encoder_update_user(uint8_t index, bool clockwise) {
    if (index == 0) {
        if (clockwise) {
            tap_code(KC_VOLU);
        } else {
            tap_code(KC_VOLD);
        }
		} else if (index == 1) {
			switch (get_highest_layer(layer_state)) {
				case _COLEMAK:
				case _QWERTY:
				case _COLEMAKDH:
					if (clockwise) {
						tap_code(KC_PGDN);
					} else {
						tap_code(KC_PGUP);
					}
				break;
			case _RAISE:
			case _LOWER:
					if (clockwise) {
						tap_code(KC_DOWN);
					} else {
						tap_code(KC_UP);
					}
				break;
			default:
					if (clockwise) {
						tap_code(KC_WH_D);
					} else {
						tap_code(KC_WH_U);
					}
				break;
		}
    }
    return true;
}
```

To:

```
bool encoder_update_user(uint8_t index, bool clockwise) {
    if (index == 0) {
        if (clockwise) {
            tap_code(KC_VOLU);
        } else {
            tap_code(KC_VOLD);
        }
		} else if (index == 1) {
			switch (get_highest_layer(layer_state)) {
				case _COLEMAK:
				case _QWERTY:
				case _COLEMAKDH:
					if (clockwise) {
						tap_code(KC_PGDN);
					} else {
						tap_code(KC_PGUP);
					}
				break;
			case _RAISE:
			case _LOWER:
					if (clockwise) {
						tap_code(KC_DOWN);
					} else {
						tap_code(KC_UP);
					}
				break;
			default:
					if (clockwise) {
						tap_code(KC_WH_D);
					} else {
						tap_code(KC_WH_U);
					}
				break;
		}
    }
    return false;
}
```

DaneEvans, 2024-03-12:
None of this is contained in this repo. Changes to these configs should be added / modified over on the QMK repo.

DaneEvans, 2024-03-12:
As such, it needs to be added there, brought up to date with current QMK, wait several months for them to do a review etc. 

I haven't touched those configs in several years, so feel free to modify them in the correct location

closed 2024-03-12 by DaneEvans