josefadamcik/SofleKeyboard #202: OLEDs not bridged Help: summarise your issue here.

gibranlp, 2024-04-03:
**Kind of issue?**
Assembly

**Which Board?**
Original  
_Please note that Keyhive is selling a customised and incompatible Sofle RGB version which is not part of this repository and therefore not supported here._

**Which Board Revision?**
V2.0

**What steps have you taken**
* [x] I have searched the Issues page
* [x] I have read the documentation 
* [x] For LED issues I have checked the following things 
        - the pro micro is programmed (and currently telling the LEDs to do something)
        - the pro micro is powered (and showing LEDs)
        - the LEDs are powered (check 5v across pins 1 and 3 of J1)
        - are any of the LEDs showing 

**What is the issue?**
Basically i soldered everything except for the bridges of the OLEDs, i was wondering if i can solder the bridges on the bottom side of the PCBs or i need to remove the pro micro to solder them on the top side.

thanks a lot

DaneEvans, 2024-04-04:
You could, but not straight across.

Assuming everything is set out as it should be, then you'll need to bridge
them over with wires, reversing the order.
  If it was just the OLED to remove, I'd say do that, but if I put them
under the micro, then definitely bridge on the other side using wires.
Those micros are a pain to remove

gibranlp, 2024-04-08:
Had to remove the Pro micros in the end, its all good now.

Thanks a lot.
Cheers.

closed 2024-04-08 by gibranlp