josefadamcik/SofleKeyboard #209: RGB model: power of led

Gabriele-tomai00, 2025-02-09:
hello, beautiful project. I have a doubt about the LEDs. I see you used leds that draw up to 60mA. The raspberry pi pico can handle 500mA maximum. How does that work? Did you reduce the brightness a lot?

JellyTitan, 2025-02-10:
I haven't tried the Sofle RGB myself, but looking at the QMK config, it appears that the default configuration for the Sofle RGB limits the maximum brightness at 50%.

[image]

If you'd like to validate that, you can see the file here:
https://github.com/qmk/qmk_firmware/blob/master/keyboards/sofle/keymaps/rgb_default/config.h

The QMK docs that refer to that setting don't go any deeper than the comment you see above:https://docs.qmk.fm/features/rgb_matrix#additional-configh-options

Gabriele-tomai00, 2025-02-10:
thank you, but even with the power halved. We have theoretically 30mA per led for almost 60 led, so 1800mA. But usb 2.0 can only deliver 500mA

DaneEvans, 2025-02-10:
Yeah, they brown out the board.  It's a balancing act.

I'm pretty sure if you read the docs it'll be mentioned.  I know I wrote
it, but it may have been for the firmware or for the CRKBD.

The real pain is when you set them too high, or set a default pattern that
is too high, so you can't change it as the mcu keeps browning out (leave
the reset header available so you can reprogram the firmware)

Gabriele-tomai00, 2025-02-10:
so you are telling me that the LEDs are programmed to always stay under a certain limit, whatever color you choose to set.
Is the brightness set this way good then? Is there any video or photos where I can see?

DaneEvans, 2025-02-11:
No, they will absolutely go bright enough to brown out the board.

They are also brighter than you need for most situations, unless you have
cases and bad light pipes.

Depending on how many LEDs you populate, what colour/s you are using, and
what brightness you want, you can set the max value to limit the chances of
exceeding what your PC will deliver.

Most will happily go over the 500mA, but how much is dependant on the
hardware.
Full power on white is basically guaranteed to brown out the keyboard
unless you only have a small number of LEDs

Gabriele-tomai00, 2025-02-11:
Thank you. I'm building a split keyboard of about 70 keys. 
I have another question, You don't use capacitors right? Are they unnecessary?

DaneEvans, 2025-02-11:
I don't recall, but I don't think so.
I used the existing designs, given that they work on this board, they are
clearly adequate.

Some capacitance may be useful, and designing from scratch I would have at
least put footprints for them (one of the reasons for the LED extension
header) but given how slowly the LEDs change in power, it's not really
an issue unless you were strobing at high power.

On top of that, PC PSUs have plenty of capacitance.  I'd only expect issues
using an unpowered hub

Gabriele-tomai00, 2025-05-29:
Hi, why did you removed the rgb version?

Dane-2pi, 2025-05-29:
Hasn't been removed so far as I can tell, and my original form isn't going anywhere. 

Look at the data sheets for the leds? Are they 5v tolerant? Or do they need something else?

Gabriele-tomai00, 2025-05-30:
ok, and It seems it's the same for the SK6803MINI-E. So what was the guide on sofle pico talking about?